MatterHackers/MatterControl #5181: Disabling the Heater Auto Shutdown feature.

MichaelPetitclerc, 2021-12-29:
Feature Request
===============

**Problem**
MatterControl's Auto Shutdown of heaters after 10 minutes is causing directly connected printers with heated chambers to stop heating (Intamsys and similar).

**Solution**
Add an option to disable this function in the "Printer" tab or the Settings menu in the top left. 

(I did not find an option to disable this feature on the current build)

larsbrubaker, 2022-01-06:
@MichaelPetitclerc can you pleas describe in detail what is happening and when one would want the heat to stay on? I want to limit this behavior as much as possible maintain this safety feature as possible. The 10 minutes happens after a cancelled print. How are you or a customer getting to this state where the printer should continue heating more than 10 minutes.

MichaelPetitclerc, 2022-01-10:
Intamsys FunMat HT's print with +300c materials, so if those materials cool in the chamber after a print and are not removed from the build plate, they will break the glass plate.  Both the Bed and Chamber heaters need to stay on until manually disabled by the end-user indicating that they cleared the print off the bed.  

At this time, the end G-Code for M140 has been adjusted to keep temp after the print.  M141 can be included in the End Code to help with the chamber temp.