MatterHackers/MatterControl #5187: Add a Bevel / Fillet operation to the 'Reshape' tools

larsbrubaker, 2022-01-13:
Feature Request
===============

### Problem
- Rounded corners and edges, which are essential in removing sharp edges from prints, need a tool that can create them for arbitrary meshes.

### Solution
- Create a tool and add it to the 'Reshape' set of tools that can bevel or round and edge

### Required Features
* When tool is added to an an object and the object is selected
  - Clicking an edge should add or remove it form the bevel set
  - Clicking another object or clicking off the object should de-select it and cancel edge selection mode
- Tool settings in the properties panel
  - Radius
  - Segments (the number of segments to make the round with)
  - Clamp overlap, limit the width of each beveled edge so that edges cannot cause overlapping intersections with other geometry.

fortsnek9348, 2022-02-24:
Duplicate issue: #5032

larsbrubaker, 2022-02-24:
@fortsnek9348 asked

Does this mean that the orthographic view should effectively put the camera at an infinite distance from the scene at all times? And how would the position of the perspective camera be determined when switching back? Remember the original position? Reposition such that the scene maintains a similar on-screen size?

______________

The new perspective position would be calculated from whatever the current orthographic perspective is. 

- A front clipping plane would be assumed based on the last interacted point from the trackball tumble widget 
- A new frustum would be created the new front clip plane and the right distance to a new eye position to simulate the view width that the orthographic view had
- The camera would then be animated in from its current near infinity position to the calculate eye position
- We would now be at the new perspective view based on the previous orthographic view

fortsnek9348, 2022-02-24:
Yes, that's what I had in mind. But I presume that this point wouldn't end up as the actual front clipping plane. It would be a "focus/reference point" and the near/far planes will remain as they normally are for the perspective projection. Unless `GetNearFar` is changed for perspective mode to update a dynamic near plane, which would need to loop over all visible triangles if not accelerated by a BVH.

larsbrubaker, 2022-02-24:
I think there could be a great value in updating the Near Far and there is both BVH and Convex hull that could accelerate the calculations. But it is not a requirement. If we can get it working well and adding a lot of value to the editing experience that is what this issue is really about.

larsbrubaker, 2022-03-04:
@fortsnek9348 Can't wait to pay you for this work. It is great

Please email me your contact info so I can send it over to our finance department. They may need to contact you for additional information.

[larsbrubaker@matterhackers.com](mailto:larsbrubaker@matterhackers.com)

closed 2022-03-04 by larsbrubaker

larsbrubaker, 2022-03-04:
Got it! Thanks.