MatterHackers/MatterControl #5189: Make library folders remember user sorting and view preferences

larsbrubaker, 2022-01-13:
- [ ] Remember the sorting and view preferences for each user
[image] [image]

jlewin, 2022-09-13:
It seems a number of revisions are needed to support this, but I'll start with the first and see if any discussion arises to clarify and confirm options:

- Library Containers need an identifier for state persistence that remains stable and unique. The most straightforward implementation is the one previously proposed, where the path (as depicted in the breadcrumb) is used as the identifier. This will allow users to dig into some arbitrary path in the file system or similar, and the system will have a unique ID to detect and/or store state. The downside to this approach is any name change along ancestor path steps will result in an ID change and loss of state.
- For persistence, the ticket suggests UserSettings so the DB sounds like the desired store.  Should the data reside in a single db record, with a string encoded json body or should each setting be stored individually with some sort of leading prefix on the key and either a hash or the raw path ID? e.g. `library-home/some/path/to/loc` or `library-123457` <- hash of string

jlewin, 2022-09-13:
If a user makes a selection, say `Icon View`, then drills into a child or parent folder, either of which have no stored view override, is it correct to presume the `Icon View` stays actively in effect until another override is encountered?

jlewin, 2022-09-13:
In addition, I think a large swath of users will deeply dislike per-folder settings, as requested by this Issue. As previously indicated, Windows Exporer has to manage these same options and it allows users to choose global or per folder state. If this feature gets added, it likely needs a companion option to allow the user to toggle between the two modes.

larsbrubaker, 2022-09-13:
Preferred implementation:
- Yes to using the full bread crumb names as the keys
- JSON in a single DB entry would be preferred
- Correct to maintain whatever view has been set when navigating into folders that don't have a view set
- I prefer the simplicity of settings per folder. But would be willing to have the option to set all subfolders to the current setting (actual implementation would be clearing settings for sub folders)

Thanks for looking at this issue and any work that you do on it.

jlewin, 2022-09-14:
Just pushed up the linked PR to track progress on this. I'm at a point where I've run out of free time but enough functionality is in place that it feels like it works, however, a number of details you spoke to above aren't yet addressed. 

- Currently each folder is persisted as its own unique setting (already had this in place before you replied). Should be easy to flip around to a single db entry with a single json blob for each folder. That has its own set of drawbacks but would only take a few minutes to put in place.
- When user level overrides are hit, they remain in effect until the next one is encountered. I think more attention is needed to make this work perfectly, but the current approach feels about right, and your pref stick until you hit a folder with differing persisted preferences.
- Opt-out option doesn't sound like it's preferred. Maybe that can wait till actual user feedback warrants a response

Feel free to merge as is or keep open until more refinements are available.

larsbrubaker, 2022-09-14:
Sorry for not understanding. The option per folder sounds like it is right. This sounds like it is all good to go.

Thank you very much for the effort.

closed 2022-09-14 by larsbrubaker

larsbrubaker, 2022-09-14:
Wow, this feels great!

I'm happy to send you this bounty if you want.

jlewin, 2022-09-14:
Great to hear. Glad it seems like an improvement and that it now gets to serve as an iteration towards the final goal. 

As for the bounty, issues marked with the tag suggest higher priority tasks and make for a good place to scan when imagining working on something.

PR #5377 (2022-09-14): Issue/5189
Possibly fixes #5189 or begins to make progress in that direction