MatterHackers/MatterControl #5232: Support interface stacking

MichaelPetitclerc, 2022-02-22:
Bug Report
============
**Steps to Reproduce**
1. Import part [Linked here](https://drive.google.com/file/d/1Ro7RDPvbFiYV7DOXMufejhaG7sJODwWI/view?usp=sharing)
2. Slice with supports enabled or add them with supports function.
3. check layer bonding. 

**Actual Behavior**
Supports are being built into the walls of the part.  The final layer of the support interface is printed above the first layer of the print, the order is correct but the physical placement is not shown correctly.  Testing with the actual print now.

[image]

**MatterControl Build Number**
2.22.02.11315

**Operating System Version**
Windows 10
**Printer Make/Model**
Pulse E-423

MichaelPetitclerc, 2022-02-23:
followup, the print went well, the virtual stacking is weird but the outcome is good support removal.

larsbrubaker, 2022-02-24:
Thanks for the bug report. It will get a test and a fix. Those lines should not print.

larsbrubaker, 2022-03-01:
Fixed. Making a new build.

closed 2022-03-01 by larsbrubaker