MatterHackers/MatterControl #5252: Port MatterControl and dependencies to .NET 5

larsbrubaker, 2022-03-03:
@larsbrubaker commented on [Thu Mar 03 2022](https://github.com/MatterHackers/MCCentral/issues/6705)

This is quite a large project and requires extensive testing.

- [ ] Port agg to .NET 5 and make sure it can pass all its tests
- [ ] Port MatterSlice to .NET 5 and make sure it can pass all its tests
- [ ] Port MatterControl to .NET 5 and make sure it can pass all its tests
- [ ] Make sure both OpenTK and GLFW are working with .NET 5

Glodigit, 2022-04-07:
I'm just curious, but why .NET 5 and not .NET 6 (and MAUI)?

larsbrubaker, 2022-04-07:
@Glodigit you make a great point. Changing this issue to .NET 6. Using MAUI is a different set of issues. The most important of which is that all we need is a window hosing service (something GLFW can provide). All of MCs UI is rendered in app and we don't use any native controls.  On top of that MAUI is still not available on LINUX so it would limit our distribution even if we wanted to take on the HUGE task of moving MC to native UI renderingi.

fortsnek9348, 2022-04-13:
Got this running on Win 10. All tests pass, except: the occasional unrelated UI timing issue (it seems), and some tests rely on the Local Library and Queue folders, which may be missing (I've marked these tests to be ignored for now). Also, .NET 6 has no AppDomains, so I kludged the test system to run some tests in a separate process. On the other hand, most UI tests now run in parallel.

Builds and runs with OpenTK for .NET framework with warnings, but any incompatibilities will be shown at runtime. Can use https://www.nuget.org/packages/OpenTK.WinForms for .NET 6, but it's pre-release and has less configurability it seems. Both appear to work. I'll probably try out OpenTK.WinForms some more. Any preference?

And is there anything else to add?

larsbrubaker, 2022-04-13:
The only thing that you didn't mention is GLFW working. You can add the attached matercontrol.json to your windows user folder and MC will run with the GLFW backend.

Other than that it sounds like the right set of features. I'll check the failing tests after the PR. 

[mattercontrol.zip](https://github.com/MatterHackers/MatterControl/files/8483083/mattercontrol.zip)

Thank you.

fortsnek9348, 2022-04-14:
GLFW appears to work for the main application, but it seems tests that don't use MatterControl's window fail because Agg UI has completely different handling when closing windows (compare `CloseSystemWindow`): `WinformsSystemWindow` will close the test window, whereas `GlfwPlatformWindow` won't. The window may either hang, or crash in a future draw event.

And some tests fail because the automation doesn't inject the shift key properly for GLFW. Different handling again.

So, nothing porting-specific so far.

PR will probably happen on or after Monday.

larsbrubaker, 2022-04-14:
That sounds great

fortsnek9348, 2022-04-21:
Further testing: I think some "timing" issues were caused by the parallel UI tests fighting over platform window focus. I've added a `static bool` to disable input from the platform.

OpenTK 4 appears to have a bug where the `GLControl` receives no keyboard input until focus leaves and returns to MatterControl. That's this bug: https://github.com/opentk/GLControl/issues/18. Setting the focus in `OpenGLSystemWindow.OnShown` appears to work. There's flexibility here though, as switching back to OpenTK 3 is just changing one value in the project.

The test `DimensionsWorkWhenNoSheet` is showing intermittent bugs still. I've seen this happen with the test in the original .NET Framework project too, so it shouldn't be caused by parallel testing. It happens once in say, 7 in 32 times. Got it captured on video. What happens is, after undoing the drag-drop resize, after the mouse clicks on the left resize widget on the cube and begins to move towards the width text edit widget ("try scaling by text entry"), the text edit control just disappears. I have no idea what causes that. Have yet to reproduce outside of testing. Commenting out the initial "Scale it wider" resize and undo code appears to stop this from happening.

fortsnek9348, 2022-04-22:
Two runs through the test suite with Open TK 4, no failures. For now.

fortsnek9348, 2022-04-23:
The keyboard input problem of OpenTK 4 GLControl seems to start with GLFW. GLFW always creates a new window, you can't attach to an existing window like OpenTK 3 does. So, Open TK 4 GLControl will have a child control that is not present in the Winforms UI hierarchy. Win32 calls are used in an attempt to prevent the control from taking focus.

It's rather inelegant. The only advantage I see over using GLFW directly, is that you don't have to translate keys, when they work.

larsbrubaker, 2022-04-29:
Bumping the bounty for this to 1,700.

larsbrubaker, 2022-04-29:
My apologies.  I saw the 'work in progress' and thought that you were not ready yet - not that I needed to review.

If you rebase on top of what is up there now, I will review them before adding any more content on top. 

[image]

Thanks for all the hard work and thanks for the heads up.

fortsnek9348, 2022-04-29:
Yes, they're drafts right now. AppVeyor needs .NET 6 to even get anywhere. And I'm sure something will pop up after that. Will rebase soon.

fortsnek9348, 2022-04-29:
And for MatterControl, AppVeyor `nuget restore MatterControl.sln` is throwing a bunch of NuGet errors as it wants precisely a `net60-windows` version of every project and package.

I've ran the command myself from https://docs.microsoft.com/en-us/nuget/consume-packages/install-use-packages-nuget-cli, and I get no errors. All I got were warnings for OpenTK 3. Is AppVeyor using an out of date tool?

I can also publish from VS to a folder without problems.

> MSBuild auto-detection: using msbuild version '15.9.21.664' from 'C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\15.0\bin'.

AppVeyor using the wrong VS version?

larsbrubaker, 2022-04-29:
I will try and get this sorted out on AppVeyor tomorrow.

larsbrubaker, 2022-04-29:
The project 'agg-sharp/agg-sharp.sln' is what AppVeyor builds and it be able to run all the samples. Without parting this we would be breaking all the agg samples and that is not something I want to do. I don't know what the scope is to get the agg samples and demos working. I'm happy to help if it is a lot.

fortsnek9348, 2022-05-13:
Okay, what's the status here. Done your testing?

larsbrubaker, 2022-05-13:
Yes. This is now considered complete and we will get payment out to you first thing next week.

Thank you for all the hard work. Along with you previous work you have really helped to make MatterControl a better project. We deeply appreciate your time and diligence.

closed 2022-05-13 by larsbrubaker

larsbrubaker, 2022-05-14:
Do you know how to attach the debugger while running tests? The new code does not allow me to set a breakpoint and it is very hard to debug what is happening with the disabled tests.

fortsnek9348, 2022-05-14:
Follow the steps in the original MatterControl PR. You need Microsoft's extension to enable child process debugging. And I've just determined that you need to disable Just My Code too.

Or if you mean to attach to a test whilst it is running, then you attach to TestInvoker.