MatterHackers/MatterControl #5295: Syndaver AXI2 Printer

HungryHam, 2022-04-02:
Could you please add the Syndaver AXI2 to the list of printer profiles.

Thanks

larsbrubaker, 2022-04-02:
If you have a working profile I would love to add it. You can export zip and post it here and I will add it to the set.

I would also be happy to send you (or anyone addressing this issuel) a  $25 gift card to MH for a good profile for this or any missing printer.

Thanks.

HungryHam, 2022-04-02:
Lars
I don't know if this what you need. It is from Sydaver AXI2 slicer Symple Slicer.
Roger

larsbrubaker, 2022-04-02:
No, it needs to be a working profile from MatterControl. It is fairly easy to set up a new machine.

[Here}(https://forums.matterhackers.com/topic/7467/creating-new-profiles) is a guide on how to do it.