MatterHackers/MatterControl #5324: Export to STL

Hal1949, 2022-05-16:
[image]

I just updated my software as suggested at the top of the application.  After the update I went back into MatterControl and finished my project and then tried to EXPORT to STL and I get the ERROR in the JPG file attached.

larsbrubaker, 2022-05-16:
Thank you for the quick response. Are you able to export the .mcx file and upload it to this issue. We would love to fix this ASAP.

We will also try to reproduce this internally and get it fixed.

Thanks again.

Hal1949, 2022-05-16:
I could save it as *.mcx file and then reload that *.mcx file.  I didn't try the other exports though.

larsbrubaker, 2022-05-16:
Can you share the MCX file with me? If you want you can email it to me larsbrubaker@matterhackers.com.

larsbrubaker, 2022-05-16:
Thanks for sharing the file. The issue has been fixed and a new build will be available soon.

closed 2022-05-16 by larsbrubaker