MatterHackers/MatterControl #5326: MatterSlice -d generates an exception, does not output settings.ini

scherej958, 2022-05-17:
Bug Report
============
**Steps to Reproduce**
1. Open command line prompt
2. type in "MatterSlice -d"
3. 

**Expected Behavior**
MatterSlice should output "settings.ini" file containing internal settings.

**Actual Behavior**
MatterSlice throw the following .NET exception:
Unhandled Exception: System.NotImplementedException: unknown type 'SEAM_PLACEMENT'
   at MatterHackers.MatterSlice.ConfigSettings.DumpSettings(String fileName)
   at MatterHackers.MatterSlice.MatterSlice.ProcessArgs(String[] args, ConfigSettings config, FffProcessor processor)
   at MatterHackers.MatterSlice.MatterSlice.ProcessArgs(String[] args)
   at MatterSlice.Program.Main(String[] args) in C:\MatterControl\MatterControl\Submodules\MatterSlice\Program.cs:line 23

**MatterControl Build Number**
MatterSlice version 2.2204
MatterControl: Test Version
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")

**Operating System Version**
Windows 10- 10.0.19043 (build 19043)
**Printer Make/Model**
FANUC / 3D Printer

scherej958, 2022-05-17:
I have temporarily worked around the issue by commenting line #505 "throw new NotImplementedException ...". I added the following lines:
lines.Add("{0}={1} # {2}{3}".FormatWith(name, value, GetEnumHelpText(property.PropertyType, property.PropertyType.Name), fieldDescription));
break;
This allows the settings.ini to output, but it outputs the issue as a commented line at the beginning of the output.

larsbrubaker, 2022-05-17:
This is now fixed in 'main'

Thank you.

closed 2022-05-17 by larsbrubaker

scherej958, 2022-05-18:
Thank you Lars for your quick response. I found the actual issue was that case: "SEAM_PLACEMENT" was missing in DumpSettings. I wanted to make sure it got into main.