MatterHackers/MatterControl #5334: USB device support

jhelmke1826, 2022-05-20:
Question When I installed Mattercontrol it updated my USB driver. My assumption it did this update to all USB ports. Is there a way to uninstall this driver? Is there a method to only have the updated driver only apply to the USB port my Pulse XE is connected to? The driver is somehow interfering with a different USB device on this computer.
========

This issue tracker is for our MatterControl development team and they are not in a position to answer support questions.

If you have a question about using MatterControl, please ask it on the MatterControl forums.
https://forums.matterhackers.com/category/20/mattercontrol

larsbrubaker, 2022-05-20:
Yes, you can uninstall the driver. 
[Here](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/using-device-manager-to-uninstall-devices-and-driver-packages) is an article from Microsoft on how to do this.

It is important to note that **we only install the driver for known printer boards** as indicated by their hardware ids.

It is also worth noting that **we do not actually install ANY driver**. We simply ask windows to use the [windows serial port driver](https://docs.microsoft.com/en-us/windows-hardware/drivers/serports/installing-plug-and-play-serial-ports-and-com-ports) for a given hardware id.

Hope that helps and you are able to resolve your issue.

jhelmke1826, 2022-05-20:
Lars,

 

With the updated firmware version for matter control and updated serial driver was also included. This updated driver works.

 

Regards,

 

Jim Helmke

 ***@***.***> ***@***.***

closed 2022-05-20 by larsbrubaker