MatterHackers/MatterControl #5358: Won't export files

kkunk, 2022-07-31:
Bug Report
============
**Steps to Reproduce**
1. click export
2. watch as nothing happens

**Expected Behavior**
open a window to save a file
**Actual Behavior**
nothing
**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
2.19.10.10370
**Operating System Version**
Linux KDE Neon (Ubuntu based)
**Printer Make/Model**
Ender 3 Pro

BuschyX, 2022-11-12:
I am having this very same problem. I am trying to export a file and it does nothing. I am using the free version and I am wondering if it is a limitation based on how big the file is or something

focus2018st, 2023-01-07:
@BuschyX  i can tell u this right now getting pro will not help, install alpha will kinda help not by much tho

larsbrubaker, 2023-01-07:
If you send me the .mcx file (export it or save to desktop or download) I would love to look at it. larsbrubaker@matterhackers.com. A screenshot or picture of what is expected and what is happening would be great too (in case I have a hard time reproducing it).

BuschyX, 2023-01-07:
I got mine working, it will export things fine now. I can't remember what it was now, but I remember there was some sort of error that popped and as soon as I cleared that error the export wasn't as issue.