MatterHackers/MatterControl #5398: corrupt exports

BuschyX, 2022-11-16:
Bug Report
============
**Steps to Reproduce**
1.  Export a design
**Expected Behavior**
open stl or gcode in another program and have it open a complete model

**Actual Behavior**
opened file renders in chunks (gcode) or needing repair (stl)

**MatterControl Build Number**
2.22.04.11498

**Operating System Version**
Win 11 pro 

**Printer Make/Model**
Aquila C2

[image]
[image]

marek-rychly, 2023-01-28:
I can confirm this issue on version 2.22.04.11498. Any ideas how to fix?

BuschyX, 2023-01-28:
> I can confirm this issue on version 2.22.04.11498. Any ideas how to fix?

I've no clue. I just export as .stl and use other programs to generate the gcode.  Also noticed that pretty much everything is exported with errors and I have to do a repair. Pain in the butt, I love this program.

marek-rychly, 2023-01-28:
Yes, geometries in exported STL are definitely corrupted. Cura slicer is able to import such an STL file, it reports warning about errors in the model (sometimes), however and fortunately, it is able to slice it into valid gcode. Prusa slicer is also able to import the STL but it is unable to slice due to the errors (more specifically, it reports open edges and wrong facets). Tinkercad (the web app) is able to import the STL but the imported object is visibly corrupted. So I am using Mattercontrol for modelling and STL export and Cura for STL import and slicing. It's a pity as for simple modelling for 3D print, Mattercontrol is the best app ever.

BuschyX, 2023-01-28:
> Yes, geometries in exported STL are definitely corrupted. Cura slicer is able to import such an STL file, it reports warning about errors in the model (sometimes), however and fortunately, it is [...]

MS 3D Builder is really good at fixing what MatterControl breaks. I usually run it through it's repair feature before loading it into my slicer program. I've tried "repairs" in other programs and while they are faster at fixing the issue, their solution is usually to make it worse, lol.