MatterHackers/MatterControl #5426: Compiling onLinux

WolfgangT71, 2023-04-03:
I use MatterControl on Linux (Debian 11)
Installed Version is 2.21.10 from 7. October.

Now I wanted to compile it again, and it does not work. Is the Version on Github already possible to compile, as it even shows Compilationfailed for windows on the github page, or does it work in Windows already?
Will the new version work in Linux as well, or are there some changes which will make it inoperable for Linux?

sjuxax, 2023-06-03:
Getting the same here.

```
❯ nuget restore MatterControl.sln
MSBuild auto-detection: using msbuild version '15.0' from '/usr/lib/mono/msbuild/15.0/bin'.
/tmp/NuGetScratch/hq9c5zk6.8p0.nugetrestore.targets(164,5): error MSB4018: The "WriteRestoreGraphTask" task failed unexpectedly. [/tmp/NuGetScratch/i7w56i0j.1av.nugetinputs.targets]
/tmp/NuGetScratch/hq9c5zk6.8p0.nugetrestore.targets(164,5): error MSB4018: NuGet.Frameworks.NuGet.Frameworks1217999.FrameworkException: Invalid framework identifier ''. [/tmp/NuGetScratch/i7w56i0j.1av.nugetinputs.targets]
/tmp/NuGetScratch/hq9c5zk6.8p0.nugetrestore.targets(164,5): error MSB4018:   at NuGet.Frameworks.NuGet.Frameworks1217999.NuGetFramework.GetShortFolderName (NuGet.Frameworks.NuGet.Frameworks1217999.IFrameworkNameProvider mappings) [0x00075] in <d1ed15ee85c14e6c83a60d66f4b60532>:0  [/tmp/NuGetScratch/i7w56i0j.1av.nugetinputs.targets]
/tmp/NuGetScratch/hq9c5zk6.8p0.nugetrestore.targets(164,5): error MSB4018:   at NuGet.Frameworks.NuGet.Frameworks1217999.NuGetFramework.GetShortFolderName () [0x00006] in <d1ed15ee85c14e6c83a60d66f4b60532>:0  [/tmp/NuGetScratch/i7w56i0j.1av.nugetinputs.targets]
/tmp/NuGetScratch/hq9c5zk6.8p0.nugetrestore.targets(164,5): error MSB4018:   at NuGet.ProjectModel.PackageSpecWriter.WriteMetadataTargetFrameworks (NuGet.RuntimeModel.IObjectWriter writer, NuGet.ProjectModel.ProjectRestoreMetadata msbuildMetadata) [0x0007d] in <d1ed15ee85c14e6c83a60d66f4b60532>:0  [/tmp/NuGetScratch/i7w56i0j.1av.nugetinputs.targets]
/tmp/NuGetScratch/hq9c5zk6.8p0.nugetrestore.targets(164,5): error MSB4018:   at NuGet.ProjectModel.PackageSpecWriter.SetMSBuildMetadata (NuGet.RuntimeModel.IObjectWriter writer, NuGet.ProjectModel.PackageSpec packageSpec) [0x00122] in <d1ed15ee85c14e6c83a60d66f4b60532>:0  [/tmp/NuGetScratch/i7w56i0j.1av.nugetinputs.targets]
/tmp/NuGetScratch/hq9c5zk6.8p0.nugetrestore.targets(164,5): error MSB4018:   at NuGet.ProjectModel.PackageSpecWriter.Write (NuGet.ProjectModel.PackageSpec packageSpec, NuGet.RuntimeModel.IObjectWriter writer, System.Boolean hashing) [0x000ce] in <d1ed15ee85c14e6c83a60d66f4b60532>:0  [/tmp/NuGetScratch/i7w56i0j.1av.nugetinputs.targets]
/tmp/NuGetScratch/hq9c5zk6.8p0.nugetrestore.targets(164,5): error MSB4018:   at NuGet.ProjectModel.DependencyGraphSpec.Write (NuGet.RuntimeModel.IObjectWriter writer, System.Boolean hashing, System.Action`3[T1,T2,T3] writeAction) [0x000a4] in <d1ed15ee85c14e6c83a60d66f4b60532>:0  [/tmp/NuGetScratch/i7w56i0j.1av.nugetinputs.targets]
/tmp/NuGetScratch/hq9c5zk6.8p0.nugetrestore.targets(164,5): error MSB4018:   at NuGet.ProjectModel.DependencyGraphSpec.Save (System.IO.Stream stream) [0x0001c] in <d1ed15ee85c14e6c83a60d66f4b60532>:0  [/tmp/NuGetScratch/i7w56i0j.1av.nugetinputs.targets]
/tmp/NuGetScratch/hq9c5zk6.8p0.nugetrestore.targets(164,5): error MSB4018:   at NuGet.ProjectModel.DependencyGraphSpec.Save (System.String path) [0x00008] in <d1ed15ee85c14e6c83a60d66f4b60532>:0  [/tmp/NuGetScratch/i7w56i0j.1av.nugetinputs.targets]
/tmp/NuGetScratch/hq9c5zk6.8p0.nugetrestore.targets(164,5): error MSB4018:   at NuGet.Build.Tasks.WriteRestoreGraphTask.Execute () [0x00106] in <d1ed15ee85c14e6c83a60d66f4b60532>:0  [/tmp/NuGetScratch/i7w56i0j.1av.nugetinputs.targets]
/tmp/NuGetScratch/hq9c5zk6.8p0.nugetrestore.targets(164,5): error MSB4018:   at Microsoft.Build.BackEnd.TaskExecutionHost.Microsoft.Build.BackEnd.ITaskExecutionHost.Execute () [0x00029] in <c70aa0e4fd934b0b9c85403668a35d13>:0  [/tmp/NuGetScratch/i7w56i0j.1av.nugetinputs.targets]
/tmp/NuGetScratch/hq9c5zk6.8p0.nugetrestore.targets(164,5): error MSB4018:   at Microsoft.Build.BackEnd.TaskBuilder.ExecuteInstantiatedTask (Microsoft.Build.BackEnd.ITaskExecutionHost taskExecutionHost, Microsoft.Build.BackEnd.Logging.TaskLoggingContext taskLoggingContext, Microsoft.Build.BackEnd.TaskHost taskHost, Microsoft.Build.BackEnd.ItemBucket bucket, Microsoft.Build.BackEnd.TaskExecutionMode howToExecuteTask) [0x002b9] in <c70aa0e4fd934b0b9c85403668a35d13>:0  [/tmp/NuGetScratch/i7w56i0j.1av.nugetinputs.targets]
```

Seems like it boils down to a missing `TargetFramework` in one of the csproj files.

krisives, 2024-05-26:
Did anyone ever have any luck?