MatterHackers/MatterControl #5436: Mouse arrow precision

alternate22, 2023-08-22:
Bug Report
============
**Steps to Reproduce**
1. In a MacBook Pro M1 Pro both the trackpad a various mice when pointing to distances (the small black cubes) or to the fields that contain values (e.g. in reshape) the pointer has to go above the expected place to work.
2. 
3. 

**Expected Behavior**
arrow to point to the targeted place

**Actual Behavior**
it requires for it to work to point several pixes above

**MatterControl Build Number**
(To find this, go to the MatterControl menu in the top left and click "About MatterControl")
2.22.04.11494
**Operating System Version**
MacOS Ventura 13.5
**Printer Make/Model**
n/a

closed 2023-08-23 by alternate22

larsbrubaker, 2023-08-24:
Thanks for updating the issue. I do know sometimes the window needs to be resized. I think there is a problem with it picking up the window boarder and factoring that into the mouse position. We'll look into it.

alternate22, 2023-08-24:
Hello again,
I think I found a pattern for this issue. When entering full screen (no top menus) the mouse behaves as expected. When exiting full screen the mouse is not properly pointing.
Best regards,
José Luis

Jose Luis Velasco
+34 638 970 560‬
***@***.***